bastibl/gr-rds #15: No module named rds_swig

St0ner1995, 2016-10-22:
when trying to run the example flowcharts, i get this error:

```
  File "/usr/local/lib/python2.7/dist-packages/rds/__init__.py", line 22, in <module>
    from rds_swig import *
ImportError: No module named rds_swig
```

when looking in that directory there are no files named rds_swig with any extension

i have tried compiling from your git repo and installing from the kali repo, neither fixed the issue

St0ner1995, 2016-10-22:
also happens when i try a custom flowchart

St0ner1995, 2016-10-22:
when reinstalling from apt, rds_swig.py exists but the error still occurs

bastibl, 2016-10-22:
It would be helpful to let us know which errors occur.

St0ner1995, 2016-10-24:
Apologies for the late reply, here is the execution log from gnuradio

```
Generating: '/root/Git/gr-rds/apps/rds_rx.py'

Executing: /usr/bin/python -u /root/Git/gr-rds/apps/rds_rx.py

linux; GNU C++ version 6.2.0 20160927; Boost_106100; UHD_003.009.005-0-unknown

Traceback (most recent call last):
  File "/root/Git/gr-rds/apps/rds_rx.py", line 39, in <module>
    import rds
  File "/usr/local/lib/python2.7/dist-packages/rds/__init__.py", line 22, in <module>
    from rds_swig import *
ImportError: No module named rds_swig

>>> Done
```

and when run from terminal

```
linux; GNU C++ version 6.2.0 20160927; Boost_106100; UHD_003.009.005-0-unknown

Traceback (most recent call last):
  File "/root/Git/gr-rds/apps/rds_rx.py", line 39, in <module>
    import rds
  File "/usr/local/lib/python2.7/dist-packages/rds/__init__.py", line 22, in <module>
    from rds_swig import *
ImportError: No module named rds_swig
```
i will try removing gnu radio and recompiling from its git repository and will let you know

St0ner1995, 2016-10-24:
using gnuradio from git fixed the issue

closed 2016-10-24 by St0ner1995

kevinselvaprasanna, 2018-01-04:
I had the same issue with GNU Radio 3.7.11  and  python 2.7.14. Setting the following environment variables fixed the issue for me:
`export PYTHONPATH=/usr/local/lib/python2.7/site-packages/
export LD_LIBRARY_PATH=/usr/local/lib`

PeGeSt, 2018-02-16:
I have the same issues under Windows:

<<< Welcome to GNU Radio Companion 3.7.11 >>>

Block paths:
	C:\Program Files\GNURadio-3.7\share\gnuradio\grc\blocks

Loading: "C:\Users\PGS\Desktop\GNU_Radio_Files\ARCHIV\RDS\Basti\gr-rds-master\apps\rds_rx.grc"
>>> Done

Loading: "C:\Users\PGS\Desktop\GNU_Radio_Files\ARCHIV\RDS\Basti\gr-rds-master_ORIGINAL\gr-rds-master\apps\rds_rx.grc"
>>> Done

Generating: 'C:\\Users\\PGS\\Desktop\\GNU_Radio_Files\\ARCHIV\\RDS\\Basti\\gr-rds-master_ORIGINAL\\gr-rds-master\\apps\\rds_rx.py'

Executing: C:\Program Files\GNURadio-3.7\gr-python27\python.exe -u C:\Users\PGS\Desktop\GNU_Radio_Files\ARCHIV\RDS\Basti\gr-rds-master_ORIGINAL\gr-rds-master\apps\rds_rx.py

Win32; Microsoft Visual C++ version 14.0; Boost_106000; UHD_003.010.001.001-0-unknown

Traceback (most recent call last):
  File "C:\Users\PGS\Desktop\GNU_Radio_Files\ARCHIV\RDS\Basti\gr-rds-master_ORIGINAL\gr-rds-master\apps\rds_rx.py", line 39, in <module>
    import rds
  File "C:\Program Files\GNURadio-3.7\gr-python27\rds\__init__.py", line 22, in <module>
    from rds_swig import *
ImportError: No module named rds_swig

>>> Done

Trying more and more since weeks and found no solution...

Any recommendations that solve the problems under Windows environment ?

Thank you very much in advance and greetings from Ranstadt / Germany

bastibl, 2018-02-16:
I don't use Windows, so I don't know about the common pitfalls there. But the problem is probably a linking error that hidden by Swig's Python boilerplate. Open `C:\Program Files\GNURadio-3.7\gr-python27\rds_init_.py` and print the errors that are thrown when trying to import rds_swig.

PeGeSt, 2018-02-18:
Hallo Basti, thank you very much for the quick reply. 
At the moment I´ve no time to try these tip - meanwhile I enjoy your great program whitout the RDS part. 
Without the RDS part:
[image]
These parts of the program runs perfect and I think you have made a great job.  I´m not fit programming python and if i have more time I ´ll try out what you  adviced to me in the above mentioned post.

Have graet day

Peter