bastibl/gr-rds #16: Block key xx not found in Platform while loadin flow graph in GRC 3.7.2.1

sulyokg, 2016-11-02:
Hi!
I successfully compiled the library sources without any errors and the libgnuradio-rds.so is installed in /usr/local/lib, but when I open apps/rds_rx.grc in GRC, it shows lots of error messages:

```
Loading: "/home/sulyokg/epg/rds/apps/rds_rx.grc"
>>> Error: Block key "gr_rds_decoder" not found in Platform - grc(GNU Radio Companion)
>>> Error: Block key "gr_rds_panel" not found in Platform - grc(GNU Radio Companion)
>>> Error: Block key "gr_rds_parser" not found in Platform - grc(GNU Radio Companion)
>>> Error: Block key "rtlsdr_source" not found in Platform - grc(GNU Radio Companion)
>>> Error: Connection between digital_diff_decoder_bb_0(0) and gr_rds_decoder_0(0) could not be made.
	sink block id "gr_rds_decoder_0" not in block ids
>>> Error: Connection between gr_rds_decoder_0(out) and gr_rds_parser_0(in) could not be made.
	source block id "gr_rds_decoder_0" not in block ids
>>> Error: Connection between gr_rds_parser_0(out) and gr_rds_panel_0(in) could not be made.
	source block id "gr_rds_parser_0" not in block ids
>>> Error: Connection between rtlsdr_source_0(0) and freq_xlating_fir_filter_xxx_0(0) could not be made.
	source block id "rtlsdr_source_0" not in block ids
>>> Done
```

Can You help Me?

bastibl, 2016-11-03:
Hi, please see the "Adapt Environment Variables" section of this tutorial
https://www.wime-project.net/installation/

sulyokg, 2016-11-03:
Hi!
Thanks for this tutorial, but this did not help, but it gave me an idea. I checked /usr/share/gnuradio/grc/blocks directory looking for the desired blocks, but they have not been found. The searched .xml files can be found at this package's grc subdirectory, so they copy to the /usr/share/gnuradio/grc/blocks directory. 
Success! Now rds_rx.grc loaded without any errors, but when I execute the flow graph, issued any python errors of PyQt5. I have python2.7 and copmiled PyQt5 from source tarball without errors, but when I execute the flow graph, the following error occurs:

```
<<< Welcome to GNU Radio Companion 3.7.2.1 >>>

Loading: "/home/sulyokg/epg/rds/apps/rds_rx.grc"
>>> Done

Showing: "/home/sulyokg/epg/rds/apps/rds_rx.grc"

Generating: "/home/sulyokg/epg/rds/apps/rds_rx.py"

Executing: "/home/sulyokg/epg/rds/apps/rds_rx.py"

linux; GNU C++ version 4.8.2; Boost_105400; UHD_003.005.005-0-unknown

Traceback (most recent call last):
  File "/home/sulyokg/epg/rds/apps/rds_rx.py", line 24, in <module>
    import rds
  File "/usr/local/lib/python2.7/dist-packages/rds/__init__.py", line 26, in <module>
    from rdspanel import *
  File "/usr/local/lib/python2.7/dist-packages/rds/rdspanel.py", line 6, in <module>
    class rdsPanel(gr.sync_block, QtWidgets.QWidget):
AttributeError: 'module' object has no attribute 'QWidget'
```

bastibl, 2016-11-03:
> Thanks for this tutorial, but this did not help, but it gave me an idea. I checked /usr/share/gnuradio/grc/blocks directory looking for the desired blocks, but they have not been found. The searched [...]

Yes, the XML files are also in the source directory, but the nicer way would be to add you install prefix to the search path of GRC, as described in the tutorial. Otherwise you will have to do something similar for every module you install.

Depending on the GNU Radio version you have installed, you might want to try the corresponding branch of gr-rds. If neither branch works, check what QT version GNU Radio is build against. If you use the next branch of gr-rds, it should be QT5.

sulyokg, 2016-11-04:
GRC has generated rds_rx.py with PyQt4 class, but another scripts uses PyQt5. I build GRC from latest sources, which uses PyQt5, but no success. Compile and build done without errors. This looks like a run-time error message generator application :(
I give up.

bastibl, 2016-11-04:
With 'latest sources', I guess you mean GNU Radio not GRC. Apart from that, the more interesting information would have been which branch you are using. If you use the master branch of GNU Radio, then please also use the master branch of gr-rds.  Same for the next branch.
Also
> GRC has generated rds_rx.py with PyQt4 class, but another scripts uses PyQt5.

and

> This looks like a run-time error message generator application :(

is not really helpful to debug.

closed 2016-11-04 by bastibl