bastibl/gr-rds #20: gr_rds_decoder, parser, panel are missing.

ziyuanjun, 2017-03-13:
I followed the README.md.
After the file rds_rx.grc loaded, i get these messages:
```
Block key "gr_rds_decoder" not found
Block key "gr_rds_panel" not found
Block key "gr_rds_parser" not found
```
Could anyone tell me what should i do to use these blocks.

bastibl, 2017-03-13:
I guess, you installed the blocks to a different directory and GRC doesn't know where the blocks are. Please adapt your `~/.gnuradio/config.conf` to include something like:

```
[grc]
global_blocks_path = /opt/local/share/gnuradio/grc/blocks
local_blocks_path = ~/usr/share/gnuradio/grc/blocks
```

ziyuanjun, 2017-03-19:
Thank you very much!  It works!
ps: Maybe due to the version i installed is _GNU Radio Companion 3.7.10_, my configure file is `~/.gnuradio/grc.conf`.

bastibl, 2017-03-19:
great!

closed 2017-03-19 by bastibl