bastibl/gr-rds #21: QT RDS Panel doesn't work

Miegl, 2017-11-20:
I tried running the Qt rdspanel.py in the next branch, but all I got was this error:
```
Traceback (most recent call last):
  File "/home/pepin/Desktop/top_block.py", line 34, in <module>
    import rds
  File "/usr/lib/python2.7/site-packages/rds/__init__.py", line 36, in <module>
    from rdspanel import *
  File "/usr/lib/python2.7/site-packages/rds/rdspanel.py", line 6, in <module>
    class rdsPanel(gr.sync_block, QtWidgets.QWidget):
AttributeError: 'module' object has no attribute 'QWidget'
```

I'm running GNU Radio Companion 3.7.11, the reason I'm trying to use the next branch is because I'm porting my application ovet to Qt.

Thanks

bastibl, 2017-11-20:
I assume you are using GNU Radio master with QT4. Could you please try the 'master' branch of gr-rds, which uses the WX GUI.
The 'next' branch (which you are using at the moment) is supposed to be used with GNU Radio's next branch.

Miegl, 2017-11-20:
Yeah I get that. I thought Qt5 can work with GNURadio 3.7.x (although other qt elemets are qt4). Is there a way to make GNURadio master 3.7 work with Qt5?
Thanks!

closed 2017-11-20 by Miegl

bastibl, 2017-11-21:
No, GNU Radio master is qt4. Either use GNU Radio master with the WX GUI of this module, i.e., use the master branch of this module with the master branch of GNU Radio. Or use GNU Radio's next branch with the qt5 GUI of this module, i.e., use next branch of this module with the next branch of GNU Radio.

Miegl, 2017-11-21:
Yeah thanks for clearing that up for me