bastibl/gr-rds #23: branch master not compatible with current gnuradio version (3.9.0.0)

CyrilZwahlen, 2019-08-22:
I've installed the latest version of gnuradio (3.9.0.0).
Unfortunately gr-rds is not compatible yet - it's still looking for gnuradio version 3.8.
Using the master branch, cmake gives me the following error:

> CMake Error at CMakeLists.txt:87 (find_package):
>   Could not find a configuration file for package "Gnuradio" that is
>   compatible with requested version "3.8".
> 
>   The following configuration files were considered but not accepted:
> 
>     /usr/local/lib/cmake/gnuradio/GnuradioConfig.cmake, version: 3.9.0.0-git

Will this be fixed soon?

bastibl, 2019-08-22:
There is no GNU Radio 3.9. It's just the development branch towards 3.9. But I'm glad you use the most recent development version. I updated cmake for the master branch. Please give it another try.

CyrilZwahlen, 2019-08-27:
Thank you very much for your work. I'm sorry to tell you that I've switched to the gnuradio branch 'maint-3.8' instead. Anyway, from the changes you've made I can understand that it would compile just fine right now ;)

closed 2019-08-27 by CyrilZwahlen