bastibl/gr-rds #24: Building documentation fails

wkazubski, 2019-10-05:
When I build gr-rds with eeabled documentation (cmake option -DENABLE_DOXYGEN=1 added), the cmake fails  with this message:
...
CMake Error: File /home/wk/rpmbuild/BUILD/gr-rds/docs/doxygen/Doxyfile.swig_doc.in does not exist.
CMake Error at /usr/lib64/cmake/gnuradio/GrSwig.cmake:58 (configure_file):
  configure_file Problem configuring file
Call Stack (most recent call first):
  /usr/lib64/cmake/gnuradio/GrSwig.cmake:112 (GR_SWIG_MAKE_DOCS)
  swig/CMakeLists.txt:40 (GR_SWIG_MAKE)

-- Configuring incomplete, errors occurred!
See also "/home/wk/rpmbuild/BUILD/gr-rds/build/CMakeFiles/CMakeOutput.log".
See also "/home/wk/rpmbuild/BUILD/gr-rds/build/CMakeFiles/CMakeError.log".

bastibl, 2019-10-06:
Thanks for reporting the issue. I just pushed some changes that hopefully fix this. Unfortunately, there is no source code documentation at this point, so I deleted the `docs` folder. The issue was caused by some cmake left overs that were not deleted.

wkazubski, 2019-10-06:
Thanks for fast respones, now cmake works OK and I can build the package (except the docs).

bastibl, 2019-10-07:
great, thanks for the feedback.

closed 2019-10-07 by bastibl