bastibl/gr-rds #25: No RDS encoding on Windows

Zanco83, 2019-11-09:
Hi,
for sure my problem is compiling related.
I tried to use rds_tx but even if everything seems to work ok (signal is correctly visible) but I think I have a problem with the RDS encoding.
I think that you should be related to the fact that that I haven't compiled your package but I don't have any error.
Where I should put the compiled libraries for the encoder?

Thanks in advance

Kind regards

bastibl, 2019-11-10:
I'm not sure what exactly you are doing. You use the package but you didn't compile or install it? I wonder why there are no errors. The flowgraph wouldn't run if blocks are missing. Anyway, before sending anything, I'd recommend to try the loopback flowgraph. It doesn't use any hardware and can be used to check if everything is installed correctly.

Zanco83, 2019-11-11:
Hi,
RDS loopback works but if connected to a real radio RDS is not decoded.
I'm using a USRP N210 with 500k of samplerate.
Do you have any suggestion?

Thanks in advance

bastibl, 2019-11-12:
AFAIS, the default flowgraph uses a sample rate of 1M. Why don't you just use it as is. You cannot just change parameters arbitrarily.

Zanco83, 2019-11-16:
Hi,
I'm not talking about signals samplerate but of USRP sink sample rate. I reduced because my pc was always frozen after start it.
Maybe you're right, the only other thing that I can share is that audio signal (a wav file) is correctly hearable on a real radio on the frequency chosen.

Thanks in advance

Zanco83, 2019-11-18:
Hi,
news from my tests, I used the rds_tx also with ubuntu but I have the same results.
Even leaving everything as it is set nothing changes.
Real radio doesn't see the RDS info, even if rds loopback works.

Thanks again

bastibl, 2019-11-19:
I can just repeat my previous answer. You cannot change parameters arbitrarily and still expect it to work. Please use the flowgraphs without modifications. The signal is designed for a given sample rate. And if this sample rate doesn't match the sample rate of the SDR is cannot work. It will still start and send something, but it's no longer standard compliant RDS.

closed 2019-12-21 by bastibl