bastibl/gr-rds #26: Module 'rds' has no attribute 'parser'

hexpwn, 2019-12-23:
When trying to run the example code I get an error. Using `gnuradio-companion` 3.8, `Python 3.8` and `gr-rds` 3.8.

`<<< Welcome to GNU Radio Companion 3.8.0.0 >>>

Block paths:
	/home/xxxx/.grc_gnuradio
	/usr/share/gnuradio/grc/blocks
	/usr/local/share/gnuradio/grc/blocks

Loading: "/home/xxxx/3/fmradio.grc"
>>> Done

Loading: "/home/xxxx/3/fmradio_2.grc"
>>> Done

Loading: "/home/xxxx/3/gr-rds/examples/rds_rx.grc"
>>> Done

Generating: '/home/xxxx/3/gr-rds/examples/rds_rx.py'

Executing: /usr/bin/python3 -u /home/xxxx/3/gr-rds/examples/rds_rx.py

/usr/lib/python3.8/site-packages/gnuradio/qtgui/range.py:41: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  precision = len(temp) if temp is not '0' else 0
Traceback (most recent call last):
  File "/home/xxxx/3/gr-rds/examples/rds_rx.py", line 586, in <module>
    main()
  File "/home/xxxx/3/gr-rds/examples/rds_rx.py", line 564, in main
    tb = top_block_cls()
  File "/home/xxxx/3/gr-rds/examples/rds_rx.py", line 139, in __init__
    self.rds_parser_0 = rds.parser(False, False, 0)
AttributeError: module 'rds' has no attribute 'parser'

>>> Done (return code 1)
`

bastibl, 2019-12-24:
I guess the shared library cannot be found (since the `LD_LIBRARY_PATH` doesn't contain the directory where the module was installed). Can you please open a Python3 console and `import rds`. If this also doesn't work search for `rds_swig.py` in the install path of the module (something like `<prefix>/lib/python3.7/site-packages/rds/rds_swig.py` and change the two `except ImportError` at the beginning to
```
except ImportError as e:
     print(e)
```

hexpwn, 2019-12-24:
I can import `rds` in Python3.

I innitially couldn't find `rds_swig.py` so I made a clean install of Gnuradio and it's packages (including `swig`). I now have the `rds_swig.py` file.

I edited the two exceptions (lines 18 and 25), adding `(...) as e:`.
```
18 except ImportError as e:
19     print(e)
20     import __builtin__
```
```
25     except __builtin__.Exception as e:
26         print(e)
27         strthis = ""
```

Still getting the same error.

bastibl, 2019-12-24:
This looks like an install problem, but it's hard to help since I don't know what you did. So here are some pointers:
- Did you check that the library path (where libgnuradio-rds.so is installed) is in your search path? It has to be in a system path (in that case, did you do `sudo ldconfig`?) or you have to set your `LD_LIBRARY_PATH` environment variable.
- Are you sure that the swig file you edited is the one that's actually used? Please add a print somewhere and check that it is printed.
- Do not debug by running through gnuradio-companion. At least in the past, it hid some problems. Just run the flowgraph from a terminal (/home/xxxx/3/gr-rds/examples/rds_rx.py).
- Did you try to import rds in Python after you edited the swig file?

hexpwn, 2019-12-24:
> Did you check that the library path (where libgnuradio-rds.so is installed) is in your search path?

It was in my search path but `gnuradio` libraries where not in the same directory. I moved `libgnuradio-rds.so` and associated files to the same dir as gnuradio and it worked!

thanks for helping me debug this. and thanks for OS'ing this project

closed 2019-12-24 by hexpwn