bastibl/gr-rds #32: RDS panel - Variable slider missing - GRC 3.8

Zanco83, 2020-04-13:
Hi,
finally I get everything to work, thanks for the previous suggestion.
I now have this configuration:
GNURadio 3.8 installed on arch linux with USRP N210.
After some tests that seems to have the correct resolution, 
I was trying to encode rds on the fly, but unfortunately variable slider blocks are missing:
Where I can find them? I add on grc.conf the grc blocks section but no blockes

bastibl, 2020-04-13:
What variable slider block? And what do you want to do with it? When I open the TX flowgraph from the 3.8 branch. There are no blocks missing.

Zanco83, 2020-04-25:
Hi,
seems that rds_tx.grc still use variable slider instead of QT GUI Range that is correctly put on rds_loopback, maybe it's just the example that needed to be updated with those.

Thanks again

closed 2020-04-26 by Zanco83