bastibl/gr-rds #33: AttributeError: module 'rds' has no attribute 'parser'

gralexey, 2020-05-30:
`Python 3.7.7` `macOS 10.14.5` 
I'm using GNURadion 3.8 from from https://github.com/ktemkin/gnuradio-for-mac-without-macports

Have installed `gr-rds` from `maint-3.8`, everything looks fine, example loads in gnuradio-companion, but I get error 
`AttributeError: module 'rds' has no attribute 'parser'`

Have tried to put `libgnuradio-rds.dylib` into ` /Applications/GNURadio.app/Contents/MacOS/usr/lib` and danced with paths
If I `import rds` in python and do `dir(rds)`, I'll get 
`['__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__', 'rdsPanel', 'rdspanel', 'unicode_literals']`

gralexey, 2020-05-30:
If I do the same for GNURadio 3.7 for `maint-3.8`, while building I get

```Undefined symbols for architecture x86_64:
  "boost::system::system_category()", referenced from:
      __GLOBAL__sub_I_decoder_impl.cc in decoder_impl.cc.o
      boost::mutex::mutex() in encoder_impl.cc.o
      boost::unique_lock<boost::mutex>::lock() in encoder_impl.cc.o
      boost::mutex::lock() in encoder_impl.cc.o
      __GLOBAL__sub_I_encoder_impl.cc in encoder_impl.cc.o
      boost::mutex::mutex() in parser_impl.cc.o
      boost::unique_lock<boost::mutex>::lock() in parser_impl.cc.o
      ...
  "boost::system::generic_category()", referenced from:
      __GLOBAL__sub_I_decoder_impl.cc in decoder_impl.cc.o
      __GLOBAL__sub_I_encoder_impl.cc in encoder_impl.cc.o
      __GLOBAL__sub_I_parser_impl.cc in parser_impl.cc.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/libgnuradio-rds.3.7.0git.dylib] Error 1
make[1]: *** [lib/CMakeFiles/gnuradio-rds.dir/all] Error 2
make: *** [all] Error 2
```

bastibl, 2020-05-31:
This is actually a question for the GNU Radio-for-mac project. I don't have a mac and I never used the project. No idea where it wants to have it's libraries installed, which version of Python you have to link against, etc etc

closed 2020-05-31 by bastibl

gralexey, 2020-05-31:
Do you know any ways I could debug this library linking issues?

bastibl, 2020-05-31:
Sorry, I have no idea how this GR installation on MAC is set up, if it's possible to compile OOTs against it, and if so which libraries to use etc. You'd have to ask the author for instructions how to use OOTs with it. Or is it just this module that doesn't work with this GR installation?