bastibl/gr-rds #41: Please add support for gnuradio-3.9

yarda, 2021-02-11:
Gnuradio 3.9 is stable. It drops swig and uses pybind, so OOTs needs some changes. There is porting guide https://wiki.gnuradio.org/index.php/GNU_Radio_3.9_OOT_Module_Porting_Guide

We are trying to rebase gnuradio to 3.9 in Fedora Rawhide.

yarda, 2021-02-24:
We used the following patch in Fedora downstream:
https://src.fedoraproject.org/rpms/gr-rds/blob/ad60e6e2b55757d6ffb47cc35dea85eb960fd100/f/gr-rds-1.1.0-gnuradio39.patch
It's untested and absolutely without any warranty :).

elafargue, 2021-04-09:
There was still a bit of work to make it actually functional, but I just opened https://github.com/bastibl/gr-rds/pull/42 which ports gr-rds to GR 3.9, feedback appreciated.

bastibl, 2021-04-14:
I just merged @elafargue's PR. There is now also a maint-3.9 branch. Please, let me know if there are any issues.

closed 2021-04-14 by bastibl