bastibl/gr-rds #47: Swig is replaced with PyBind in GR 3.9

solomonbstoner, 2021-05-27:
Got the following compilation error while running `cmake ..` . My GR is compiled from source from the gnuradio master branch as of commit 39af5fb4ad9cce452406f905039af4349a6ef828. Swig has been dropped in support of PyBind.

```
CMake Error at swig/CMakeLists.txt:26 (include):
  include could not find load file:

    GrSwig

CMake Error at swig/CMakeLists.txt:37 (GR_SWIG_MAKE):
  Unknown CMake command "GR_SWIG_MAKE".
```

solomonbstoner, 2021-05-27:
You know what? I was the idiot here. I didnt check the branch. The default branch is `maint-3.8`. I assumed it was `master`. The `master` branch is fine. Thanks @bastibl

closed 2021-05-27 by solomonbstoner

balistudies, 2023-03-01:
where exactly i have to change the branch, am newbie

bastibl, 2023-03-01:
please look for a git tutorial.

balistudies, 2023-03-01:
thanks, such a nice reply. looking on this issues, it exist for years

bastibl, 2023-03-02:
You don't seem to get the idea of an issue tracker. This issue was opened and closed nearly immediately since the author realized he used the wrong branch. It will not be delete (so it will "exists" for many more years), hoping that is will be found and useful for others, making the same mistake. Didn't work for you obviously. But independent from that your comment is just ridiculous.

balistudies, 2023-03-10:
an issue tracker is for the developer, to be informed about an issue, so the developer can fix it, isn't.

your response is rude and impolite, anyway, i made it with linux.
friendly advice, rather to put something in the public, i would close it or best option is to fix it.
i think, its human way and shows your ethic. i guess its german way too as to deliver a full working product.
you have a phd, i learnt a different approch, but academic isn't = academic.
alles gute basti!

bastibl, 2023-03-10:
look, I find it rude that you think it is fine stealing my time, expecting that I would explain you how to use git, even though it's trivial to google it.