bastibl/gr-rds #48: Runtime Error on Fedora 34

dklann, 2021-09-15:
Summary: successful build and install (to default _/usr/local_) on Fedora 34 using repository versions of gnuradio, osmosdr, etc (and their _devel_ equivalents), I get the following error:
```
Traceback (most recent call last):
File "/usr/local/lib64/python3.9/site-packages/rds/__init__.py", line 17, in <module>
from .bindings.rds_python import *
ModuleNotFoundError: No module named 'rds.bindings'
```
I'm not very fluent with Python, and tried to track this down, but it eludes me.

Any suggestions on where to look for this module?

Thanks!

bastibl, 2021-09-15:
How did you install GNU Radio and which version? *devel* suggests that you used the package manager. Installing  to `/usr/local` suggests that you compiled from source.

The version of the OOT that you are using (you use the `maint-3.9` branch) is for GNU Radio 3.9. Make sure that GNU Radio 3.9 is installed.

If dependencies are fine, rebuild the module and check the cmake output closely to make sure it found all dependencies and created all bindings.

If this worked, make sure that the Python file that it is looking for exists? Check `/usr/local/lib64/python3.9/site-packages/rds`.

If it exists, the problem might be that a shared library cannot be found. Try `sudo ldconfig`. And then check if everything is found by calling `ldd <.so file in /usr/local/lib64/python3.9/site-packages/rds>`. If the libraries cannot be found, you have to adapt your `LD_LIBRARY_PATH`.

dklann, 2021-09-15:
Thanks for your quick reply @bastibl.

I installed gnuradio from the Fedora repository, version 3.9.0.0. Here is the output of `gnuradio-config-info`:
```
/usr
/etc
/etc/gnuradio/conf.d
/home/dklann/.gnuradio
Wed, 17 Feb 2021 00:00:00
testing-support;python-support;post-install;doxygen;gnuradio-runtime;gr-ctrlport
3.9.0.0
gcc (GCC) 11.0.0 20210210 (Red Hat 11.0.0-0)
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software see the source for copying conditions.  There is NO
warranty not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
g++ (GCC) 11.0.0 20210210 (Red Hat 11.0.0-0)
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software see the source for copying conditions.  There is NO
warranty not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
/usr/bin/gcc:::-DNDEBUG -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m64  -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fvisibility=hidden -Wsign-compare -Wno-uninitialized
/usr/bin/g++:::-DNDEBUG -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Werror=format-security -Wp,-D_FORTIFY_SOURCE=2 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m64  -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fvisibility=hidden -Wsign-compare -Wno-uninitialized
```
_/usr/local_ is the location of the installed _gr-rds_ files.

The only _cmake_ error displayed is this:
```
-- Checking for module 'mpir >= 3.0'
--   Package 'mpir', required by 'virtual:world', not found
```
Even though the Fedora packages _mpir-3.0.0-15.fc34.x86_64_, _mpir-devel-3.0.0-15.fc34.x86_64_, and _mpir-c++-3.0.0-15.fc34.x86_64_ are installed from the Fedora repository. Full output from cmake is at https://pastebin.com/TBwJLjCa

Even with that error, _gr-rds_ installs as expected. I have run `sudo ldconfig` and ensured that _/usr/local/lib64_ is included in the list of "local" directories. The output of `ldd /usr/local/lib64/python3.9/site-packages/rds/rds_python.cpython-39-x86_64-linux-gnu.so` shows no missing libraries. The complete output of `ldd` is at https://pastebin.com/L2a2K23h and the full output of `sudo make install` is at https://pastebin.com/hGTqbDHc

With all that, I am still seeing the following error (I am including the complete output this time):
```
Traceback (most recent call last):
File "/usr/local/lib64/python3.9/site-packages/rds/__init__.py", line 17, in <module>
from .bindings.rds_python import *
ModuleNotFoundError: No module named 'rds.bindings'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
File "/home/dklann/src/gr-rds/examples/rds_rx.py", line 45, in <module>
import rds
File "/usr/local/lib64/python3.9/site-packages/rds/__init__.py", line 19, in <module>
from .rds_python import *
ImportError: generic_type: type "decoder" referenced unknown base type "gr::sync_block"
```

bastibl, 2021-09-15:
This seems to be related to pybind versions: https://www.mail-archive.com/discuss-gnuradio@gnu.org/msg73247.html

dklann, 2021-09-15:
Well, that likely explains things. Fedora 34 ships (and I had) _pybind11-devel-2.6.2-1.fc34.x86_64_ and _python3-pybind11-2.6.2-1.fc34.x86_64_.

I uninstalled those packages and installed the current pybind11 using the command `sudo python3 -m pip install /usr/lib/python3.9/site-packages pybind11`. This resulted in pybind11 version 2.7.1 installed. But now `cmake ..` doesn't find a pybind11 at all:
```
CMake Error at CMakeLists.txt:123 (find_package):
By not providing "Findpybind11.cmake" in CMAKE_MODULE_PATH this project has
asked CMake to find a package configuration file provided by "pybind11",
but CMake did not find one.

Could not find a package configuration file provided by "pybind11" with any
of the following names:

pybind11Config.cmake
pybind11-config.cmake

Add the installation prefix of "pybind11" to CMAKE_PREFIX_PATH or set
"pybind11_DIR" to a directory containing one of the above files.  If
"pybind11" provides a separate development package or SDK, be sure it has
been installed.

-- Configuring incomplete, errors occurred!
```
And there is no _Findpybind11.cmake_ with the pip-installed version of pybind11.

I'm in way over my head at this point.... Is it more appropriate to move this to the _discuss-gnuradio_ mailing list?

bastibl, 2021-09-15:
You probably only installed the Python parts of pybind11. My pybind11 dev package has quite a lot of headers, etc. Either you don't have them or pip might have installed them at a place where GNU Radio doesn't know about.

```
basti@thrang ~ >> dpkg -L pybind11-dev
/.
/usr
/usr/include
/usr/include/pybind11
/usr/include/pybind11/attr.h
/usr/include/pybind11/buffer_info.h
/usr/include/pybind11/cast.h
/usr/include/pybind11/chrono.h
/usr/include/pybind11/common.h
/usr/include/pybind11/complex.h
/usr/include/pybind11/detail
/usr/include/pybind11/detail/class.h
/usr/include/pybind11/detail/common.h
/usr/include/pybind11/detail/descr.h
/usr/include/pybind11/detail/init.h
/usr/include/pybind11/detail/internals.h
/usr/include/pybind11/detail/typeid.h
/usr/include/pybind11/eigen.h
/usr/include/pybind11/embed.h
/usr/include/pybind11/eval.h
/usr/include/pybind11/functional.h
/usr/include/pybind11/iostream.h
/usr/include/pybind11/numpy.h
/usr/include/pybind11/operators.h
/usr/include/pybind11/options.h
/usr/include/pybind11/pybind11.h
/usr/include/pybind11/pytypes.h
/usr/include/pybind11/stl.h
/usr/include/pybind11/stl_bind.h
/usr/lib
/usr/lib/cmake
/usr/lib/cmake/pybind11
/usr/lib/cmake/pybind11/FindPythonLibsNew.cmake
/usr/lib/cmake/pybind11/pybind11Common.cmake
/usr/lib/cmake/pybind11/pybind11Config.cmake
/usr/lib/cmake/pybind11/pybind11ConfigVersion.cmake
/usr/lib/cmake/pybind11/pybind11NewTools.cmake
/usr/lib/cmake/pybind11/pybind11Targets.cmake
/usr/lib/cmake/pybind11/pybind11Tools.cmake
/usr/share
/usr/share/doc
/usr/share/doc/pybind11-dev
/usr/share/doc/pybind11-dev/changelog.Debian.gz
/usr/share/doc/pybind11-dev/copyright
```

dklann, 2021-09-16:
That was indeed the case. I manually linked the _include_ directory and the _lib/cmake_ directory to their standard locations on Fedora 34:
```
sudo ln -s /usr/local/lib/python3.9/site-packages/pybind11/include/pybind11 /usr/include
sudo ln -s /usr/local/lib/python3.9/site-packages/pybind11/share/cmake/pybind11 /usr/lib64/cmake
```
and rebuilt from scratch. CMake completed without error and running `python3 examples/rds_rx.py` results in the same `ModuleNotFoundError: No module named 'rds.bindings'`

This whole adventure is puzzling! It's seems weird to me that _gnuradio_ and _python3-gnuradio_ (3.9.0.0-5) installed from the Fedora repository would have been compiled with a different version of _pybind11-devel_ (2.6.2-1) that is in the same repository. Is it possible that there is some behavior in pybind11 that gr-rds depends on?

Signed, "Very Confused"

bastibl, 2021-09-16:
Not that I would know of. I'd recommend to just try any other 3.9-compatible module (that's not from me) and check if it makes a difference. https://www.cgran.org/
If it's specific to this or my modules, we can continue here. Otherwise it might make sense to ask on the GNU Radio mailing list. I don't use Fedora, so there might be something obvious that I'm missing.

dklann, 2021-09-16:
Your hunch (if that's what it was) is correct @bastibl. This is quite certainly a Fedora and pybind11 issue. I just installed gr-satellites, and running one of its binaries `gr_satellites` produces the error _ModuleNotFoundError: No module named 'satellites.bindings'_

I'll take this up with the gnuradio folks on discuss-gnuradio.

Thanks again for your help in troubleshooting this!

closed 2021-09-16 by dklann