bastibl/gr-rds #55: GR_SWIG_MAKE_DOCS kali linux GNU radio 3.8.2.0 (Python 3.9.1)

pierregalant, 2021-11-25:
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.9.so (found version "3.9.1") 
CMake Error: File /root/gr-foo/docs/doxygen/Doxyfile.swig_doc.in does not exist.
CMake Error at /usr/local/lib/cmake/gnuradio/GrSwig.cmake:58 (configure_file):
  configure_file Problem configuring file                                                                                                                                                                                                  
Call Stack (most recent call first):                                                                                                                                                                                                       
  /usr/local/lib/cmake/gnuradio/GrSwig.cmake:112 (GR_SWIG_MAKE_DOCS)                                                                                                                                                                       
  swig/CMakeLists.txt:50 (GR_SWIG_MAKE)

pierregalant, 2021-11-25:
Hello, I am having this error when install gr-foo on maint-3.8. My pc is running kali linux with radio 3.8.2.0 (Python 3.9.1)

bastibl, 2021-11-25:
You open an issue with the exact same problem also in gr-ieee802-11. I'll close this one.

closed 2021-11-25 by bastibl