bastibl/gr-rds #63: Import RDS Fails

CanBusHack, 2022-05-04:
Loading the RDS Module fails after syncing, gnuradio/gr-osmocomsdr/gr-rds to versions 3.10.1 (not sure why it's looking for 3.10.2git).  

Install gnuradio from gnuradio ppa (ppa:gnuradio/gnuradio-releases).
Installed gr-osmocomsdr from git using branch "master"
installed gr-rds (branch maint-3.10)

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/rds/__init__.py", line 17, in <module>
    from .bindings.rds_python import *
ModuleNotFoundError: No module named 'rds.bindings'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/user/git/gr-rds/examples/rds_rx.py", line 44, in <module>
    import rds
  File "/usr/local/lib/python3.9/dist-packages/rds/__init__.py", line 19, in <module>
    from .rds_python import *
ImportError: libgnuradio-runtime.so.3.10.2git: cannot open shared object file: No such file or directory

bastibl, 2022-05-05:
You still either have artifacts from previous installation attempts. Or if you actually wanted to link against 3.10.2.git., the library is not in the library search path of your linker (-> `LD_LIBRARY_PATH`).

CanBusHack, 2022-05-05:
I do not nor have I ever had gnuradio 3.10.2git on my computer. Not sure why it is trying to find this. 

I wanted to use gnuradio 3.10.1.1 but "from .rds_python import *" must have some reference to this version and I'm not sure how it got there.

bastibl, 2022-05-05:
I don't know what you are doing. But the reference to the library is clearly not from this module. But anyway, your other issue looks like you started from scratch.

closed 2022-05-05 by bastibl