bastibl/gr-rds #70: rds in

on4akh, 2023-02-04:
Hello,

Would it be possible to give an example to use the rds in port on the RDS encoder block ?
Looking the code it seems possible to set for example the radio text.
But I can not get it to work.

Thanks
Wim.

sterlan, 2023-02-04:
Hello Wim. 
Why do you need the rds_in port of the RDS Encoder block? I think the rds data is set to the RDS Encoder(Radiotext, PS which represents the name station et.c). Correct me someone if i'm wrong.
Stergios.

on4akh, 2023-02-05:
Hello Stergios,
Thanks for your quick reply. I would like to change the radio text remotely.
Anyways, in the mean time I have found how to do it. I added a PDU Socket block to the rds in port. To change the radio text you have to send 'text<your text>' so for example textHello. The update however is kind of slow. Takes about a minute or so to see the actual change on a receiver.
Wim.

closed 2023-02-05 by on4akh