bastibl/gr-rds #72: Small comments to the great .grc examples

Brumi-2021, 2023-06-24:
Hello thanks a lot for sharing that gr-rds  and FM stereo ( rx / tx) . 
Great job,  it is the perfect reference that I was searching,  and also it works  with version gnuradio 3.10.6  ! 

Just let me add small comments to your examples, for your consideration , 
(1) [rds_rx.grc](https://github.com/bastibl/gr-rds/blob/maint-3.9/examples/rds_rx.grc)
Regarding that deemphasis block , as I am in EUR,   I would just add this comment ,
Tau : Time constant in seconds (75us in US, 50us in EUR)
[image]

(2) [rds_tx.grc](https://github.com/bastibl/gr-rds/blob/maint-3.9/examples/rds_tx.grc)
Here in that transmitter example , I would just add two comments ,

a-) I think it is missing to add in audio baseband , the two symetrical pre-emphasis filters , 
[image]

b-) Regarding the modulation index level of each part ,  If I understood correctly , current grc is using 26% for (L+R) and (L-R),
[image]
(but not sure if this is the right FM stereo specs) 

Then , based on that , my understanding is that in FM STEREO transmission,  
(L+R) baseband ,  and (L-R ) double sideband suppressed carrier (DSBSC) sub-channel , could be each one 42,5%,  
Pilot Tone 19Khz 10% , and RDS BPSK around  5% .

[image]

In that sense , I think we can increase that 0.26  to 0.425 , if I am not wrong . 

Feel free to close or modify that issue , if it is not completely correct, 

Cheers, and thanks for your great contribution to gnuradio community !!!

bastibl, 2023-06-24:
Thank you very much for the issue and for the detailed comments. This sounds very reasonable to me. Would you mind opening a PR to incorporate these changes?

Brumi-2021, 2023-06-24:
Ok , thanks , I will try to submit the related small PR to update accordingly those two GRC examples . 

BTW , two more small comments , in the current receiver , I do not see two correct dots in the constellation, I saw an horizontal line , maybe it is my set up problem . 
And regarding the transmitter , I tried several different receivers . The stereo separation works very good ,  but I always can hear some small residual background audio noise there in both channels , not sure where it comes from , pending to investigate it . I tried to increase the RF signal but not related. Anyone had same issue ?

bastibl, 2023-06-26:
Closed with https://github.com/bastibl/gr-rds/pull/73

closed 2023-06-26 by bastibl