bastibl/gr-rds #75: Receiving TMC question

THuber2145, 2023-08-07:
Hello,

thanks for your awesome Project.

I have a question about receiving tmc messages, i have the latest version of your blocks and your rx example (grc 3.10) up and running with an RTL-SDR dongle. RDS works great!

I may be blind, but where can I see the output of the decoded tmc messages?
Sorry to bother you, keep up the good work!

bastibl, 2023-08-07:
Hi, not all stations broadcast TMC message all the time. If TMC messages are decoded, the *RDS Parser* block should print them to standard out, if logging is enabled.
See here: https://github.com/bastibl/gr-rds/blob/maint-3.9/lib/parser_impl.cc#L405-L414 

You can enable logging in the options of the *RDS Parser* block and try different radio stations.

THuber2145, 2023-08-07:
Hi, thanks for the  for the extremely quick response, 
I'll take a look at that parser block again!

closed 2023-08-07 by THuber2145