bastibl/gr-rds #76: Field order in RDSPanel

marshallartsoz, 2024-05-05:
This is all brilliant stuff, thanks so much for putting in the work, and for making it available. I have installed the maint-3.10 branch on my Linux Mint system, and it works great.  Just one odd thing I noted - in the window that comes up when I run the rds_rx.grc flow in the GnuRadio Companion, the fields for Program ID and Program Type seem to be the wrong way around.  I'm not a Python coder, but I had a look at the rdspanel.py code, and nothing hit me in the eye that looked wrong.  Not a big deal obviously, but I just like things to look right. I've attached a clip of a screenshot that shows what I mean.

[image]

bastibl, 2024-05-06:
Can you check the `rdspanel.py` file of your zip file, if it has the changes from #71? Maybe there is also an older gr-rds installation from your package manager (or similar) on your system, which also installed the RDS panel and this gets used when you start the flowgraph.

marshallartsoz, 2024-05-06:
It does now.  Thanks.  I'm not exactly clear on what was happening there, but no matter, it's fine now.

closed 2024-05-06 by marshallartsoz