bastibl/gr-rds #77: Following the instructions, fails to compile. "Missing GnuradioConfig.cmake, gnuradio-config.cmake"

JohnCC330, 2024-07-16:
Executing the compilation recipe:
```
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig
```
Fails at the `cmake ..` step:
```
CMake Error at CMakeLists.txt:77 (find_package):
  By not providing "FindGnuradio.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Gnuradio",
  but CMake did not find one.
```
I'm not familiar with the cmake process...

bastibl, 2024-07-17:
Make sure that you have GNU Radio installed (with development files).

JohnCC330, 2024-07-19:
On Wed, 17 Jul 2024 05:42:30 -0700
Bastian Bloessl ***@***.***> wrote:

> Make sure that you have GNU Radio installed (with development files).

Hi Bastian,

Gnu radio _is_ installed - I use it regularly. I compiled and installed it,
including headers.

John

bastibl, 2024-07-19:
When installing GNU Radio from source, you might have to adapt environment variables to make sure that libraries, python libraries, pkg config/cmake is found.
Usually, if you use the same install path (`-DCMAKE_INSTALL_PREFIX`), cmake files should be found.
Also make sure that you use the version, corresponding to the branch of this module.

JohnCC330, 2024-07-20:
Slackware is a little different from other distributions, in that, by default, files are installed in /usr/... particularly, libraries are installed in /usr/lib64 and include files in /usr/include. The LD_LIBRARY_PATH etc. of course include those. Only very rarely do I have problems...

bastibl, 2024-07-23:
When you have an unusual system, you might have to set up more environment variables to make sure everything is found. Since it complaining that it does not find `FindGnuradio.cmake`, I'd suggest to look if this file was installed and try adding it manually to the `CMAKE_MODULE_PATH`, for example, by changing the `CMakeLists.txt` file. Just add a line, similar to this: https://github.com/bastibl/gr-rds/blob/maint-3.10/CMakeLists.txt#L30
Maybe the Slackware or cmake people might be able to help you better with this.

closed 2024-10-28 by bastibl