bastibl/gr-rds #82: RDS isn't generated/TEF6686 doesn't show RDS flag

gabba43, 2024-10-27:
Hi everyone, first off: Awesome project! I really appreciate your work. My hobby is rf engineering & fm broadcasting and I'm currently trying to build a link receiver and mpx (RDS+stereo) encoder on a Raspberry Pi 2 W or Orange Pi Zero 3.

Now the issue I faced: Your RDS encoder is not working. Tried it on a Linux Mint 12 desktop with a VCO-PLL-tx by PCS Electronics. My 384kHz Realtek sound card is not the issue as it works perfectly fine on the same system using Stereo Tool. Same on a Win10 machine with Airomate: The sound card does generate proper stereo & RDS.

Tried the branch for GnuRadio 3.9 as well as the new version for GnuRadio 3.10 - no luck. QT Frequency FFT shows a correctly generated 57kHz-RDS-subcarrier & 19kHz stereo pilot. Also, I get a sound output on that sound card in GnuRadio - but no RDS.

Checked everything using my Tef6686 - the RDS flag doesn't light up. When using StereoTool or Airomate, it does immediately.

Do you know what's the issue here? Maybe I made a mistake.

bastibl, 2024-10-27:
Could you please upload the modified TX flowgraph some where please?

argilo, 2024-10-28:
The sample rate at the output of the Add block is 380 ksps, so you can't directly attach a 192 ksps Audio Sink and expect it to work. You would need to add a Rational Resampler block in between.

argilo, 2024-10-28:
Interpolation 192, Decimation 380.

gabba43, 2024-10-28:
Thank you very much for your fast reply! I'll give it a try.

gabba43, 2024-10-28:
I'm jumping in circles with joy :D It works! You're a genius. When I finished my project with the Orange Pi, I'll tell you and post everything on Radionecks - of course crediting you!

closed 2024-10-28 by gabba43