bastibl/gr-rds #83: Running rds_tx.grc/.py headless on ARM64 Orange Pi?

gabba43, 2024-10-28:
Sorry to bother you again :)

now, I installed gnuradio on my Orange Pi Zero 3 and tried to run the script headless by changing it to 'No GUI' in the options panel and exporting it to a .py-file. It certainly runs on the OPi and doesn't give any other errors than on my Linux Mint machine but there's no MPX generated. On my desktop machine the .py-script runs perfectly fine! Any help would be greatly appreciated!

System: OPi Zero 3, 2 Gb ram, Allwinner H618/Debian bookworm server, 6.1 kernel, ARM64; version 1.0.4 (Orange Pi release)

gabba43, 2024-10-28:
[rds_tx_for_opi.zip](https://github.com/user-attachments/files/17548639/rds_tx_for_opi.zip)

bastibl, 2024-10-28:
You seem to specify the audio devices manually (hw:4,1 and hw:3,0). Maybe the mapping is different on the different platforms, the devices support different sample rates, or you want to use a different audio subsystem (e.g. pulse)?

gabba43, 2024-10-28:
Thank you! Forgot to say that. I saved the installation logs of both gnuradio & deps as well as gr-rds, I can post them tomorrow. Nonetheless there were no errors when compiling & installing.

Also, I uninstalled & reinstalled gnuradio & deps after the first try but changed the repos to the official debian ones. Maybe that helps.

I made sure that the mapping & sample rates are correct but I'll check tomorrow again. Maybe I'll add an entry to .asoundrc to make sure the USB-sound card is fixed to 192kHz. I deactivated pulse on the desktop beforehand already as I'm using jack (which only runs once started) - so when not using jack, only ALSA is running. On the debian server, I think there's also only ALSA installed but I'll make sure that this is the case tomorrow. Maybe it has pipewire or so, I'll kick that out.

gabba43, 2024-10-29:
So, after uninstalling, changing deps to official ones and reinstalling, it finally works! But... CPU load is on 95% on the Orange Pi Zero 3, 2gb. RAM usage isn't a problem, less than 100mb. RDS pops up and goes off again. So the OPi is underpowered. Luckily, I still have a couple of HP 260 G2 DMs with Pentium 4405U CPUs laying aroung which will be perfect for this job (I hope). I'll install a headless Debian on them as well. Now I just need to get the dynamic PS (changing every 3 secs) to work :D

bastibl, 2024-10-29:
I just tested it and it really takes much more CPU than I would have expected. That's in no way optimized :-/ I hope, it works with your other device.

argilo, 2024-10-30:
A bunch of blocks have Maxoutbuf set to 10, which forces samples to be processed in very small batches. This seems to increase CPU load a lot.

Unfortunately, removing that results in a lot more output buffer underruns. The Audio Source block has "OK to Block" set to "No" to avoid the two-clock problem, but unfortunately that setting is ignored in nearly all audio backends (except Portaudio, which is unlikely to be used because ALSA has a higher priority).

gabba43, 2024-10-30:
I understand. So setting a fixed, higher, buffer size and inputting audio with a fixed (e.g. 4096), higher, buffer size (-bufsize?) using ffmpeg is not a solution?

gabba43, 2024-10-30:
Awesome :D Actually tried that already but changed only the sample rate which didn't work out.

argilo, 2024-10-30:
One small correction (which I edited in above): Root Raised Cosine Filter interpolation should be 16.

gabba43, 2024-10-30:
Boomshakalaka! It works like a charm! CPU load is at about 5-8% when playing & resampling audio to 48kHz in FFMPEG and encoding MPX (stereo & RDS) at the same time. I'm gonna finish my tutorial and post it today, crediting you of course. I'll post a link here as well. Also, I added both of my project files, the .grc & the .py. Thank you very much again!

argilo, 2024-10-30:
One last thing: from the screenshot above, it looks like you weren't using the latest version of the rds_tx.grc flow graph. You'll want to incorporate the recent changes made in #79, otherwise FM stereo channel is not going to work properly with the 192 ksps intermediate sample rate. (The left & right channels may not remain separated.)

gabba43, 2024-10-30:
Here's that one with latest version of the rds_tx.grc flow graph.
[rds_tx_for_opi_1.0.zip](https://github.com/user-attachments/files/17579119/rds_tx_for_opi_1.0.zip)

argilo, 2024-10-31:
Nice to see you got it working!

It should be possible to do dynamic RDS updates via TCP if you put back the Socket PDU block and connect it to the RDS Encoder. Some details in this issue: #44

Latency is probably going to be an issue, since a lot of RDS bits will probably build up in GNU Radio buffers. gr-rds should probably have some latency management in the style of gr-latency_manager. References:

* https://www.youtube.com/watch?v=jq0RewceCwc
* https://github.com/dkozel/gr-latency_manager

argilo, 2024-10-31:
The changes I proposed in #84 are:

* Multiply, Add, Frequency Mod: change Maxoutbuf from `outbuffer` to 0
* RDS Encoder, Chunks to Symbols: change Maxoutbuf from 0 to `outbuffer`

If you make those changes in your flow graph (apart from the Frequency Mod block, which your flow graph doesn't have) you should be able to set the `outbuffer` variable back to 10 to keep RDS latency under control.

There will still be 20 seconds of RDS latency, which is not ideal, but it's much better than three minutes, which you'll get with the current flow graph if you set `outbuffer` to 0.

gabba43, 2024-11-08:
Sorry for my late reply, here's the updated version:
[rds_tx_for_opi_1.1.zip](https://github.com/user-attachments/files/17682623/rds_tx_for_opi_1.1.zip)

argilo, 2024-11-08:
Looks good.

By the way, you could shave another 7 seconds or so off the latency by manually editing the `.py` file to replace:

```
self.digital_chunks_to_symbols_xx_0.set_max_output_buffer(10)
```

with

```
self.digital_chunks_to_symbols_xx_0.set_max_output_buffer(0, 10)
```

This is a workaround for this bug: https://github.com/gnuradio/gnuradio/issues/7534

gabba43, 2024-12-27:
I got good news. After some time, I managed to get the dynamic RDS going. Here's my python code for the socket PDU client: 

import socket
import time

HOST = 'localhost'  # oder die IP-Adresse des Servers
PORT = 52001        # Port des Socket PDU Blocks in GNU Radio

s = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
s.connect((HOST, PORT))

try:
    while True:
        s.send(b"ps Radio\n")
        time.sleep(3)
        s.send(b"ps Berlin\n")
        time.sleep(3)
except KeyboardInterrupt:
    s.close()

[rds_tx_for_opi_1.2.zip](https://github.com/user-attachments/files/18262332/rds_tx_for_opi_1.2.zip)

gabba43, 2025-04-13:
I have one last question before we can actually close this issue. 

Could you please explain to me, how I can generate a stereo-signal that consists of the normal L/R-stereo-channels AND only the RDS-signal on top of that (no MPX/preemphasis). 

This would be very useful to inject RDS into those small chinese transmitters that use FM-ICs. I'm succesfully doing that already using for example Airomate or StereoTool - you can do the mixing of L/R & RDS digitally or analogue, both works. It would be cool to have it one application though.

I tried it already once by reducing the flowgraph/eliminating the unnecessary nodes but it didn't work out for me. I assume that I need to change the Multiply & Add additionally - also asked ChatGPT for help but couldn't figure it out.

Many thanks for your help!
gabba43

argilo, 2025-04-14:
> This would be very useful to inject RDS into those small chinese transmitters that use FM-ICs.

Could you show us what type of transmitter you're talking about? What does it expect to receive as input?

argilo, 2025-04-17:
It could certainly be done, but many adjustments would have to be made to the example rds_tx.grc flow graph, and you'd have to use a sound card with support for 192 kHz output.

I'm not inclined to spend time on it though, since the resulting RDS signal quality seems to be low, and the gr-rds project is intended for use with SDR transmitters that can produce much cleaner output.

gabba43, 2025-04-17:
Thank you for your reply! I have multiple 192/384 kHz soundcards laying around. Maybe I'll give it a try in the future. 

Btw, so far, using gr-rds as an MPX encoder on the Orange Pi 3 works like a charm.

For now, my plan is to record Airomate's RDS output to a .wav-file and mix it together with the webstream on my Linux Mint machine using a simple dmix device.

Did you check your SDR's (Hackrf?) spectral purity on a spectrum analyzer? My chinese 120w FM-transmitter based off the BH1418 doesn't interfere with any other FM station - even weaker ones. I was positively surprised about it's spectral purity on the spectrum analyzer. AFAIK it's certainly a lot cleaner than those SDRs.

argilo, 2025-04-17:
I wasn't thinking of the overall spectral purity (external filtering would be needed to remove harmonics from most general-purpose SDRs) but rather the stated limitations of the MP3 transmitters. From the pica.cz document:

> Internal limiter and stereo encoder causes considerable reduction of RDS signal quality injected that way.  Pilot and RDS synchronization is omitted. Due to poor quality of the integrated stereo encoder there's a strong 3rd pilot harmonic at 57 kHz, interfering with the RDS signal.

An SDR transmitter would not have these issues.

mickeyreg, 2025-06-09:
I found in this thread useful code to control RDS Encoder by sending data to TCP socket, send by @gabba43. Based on this code I wrote (with help of ChatGPT) an Embedded Python Block that can be used to change RDS data from GRC. I think someone might find this code useful, so I decided to share it in this thread:

[test_of_rds_blocks.zip](https://github.com/user-attachments/files/20656660/test_of_rds_blocks.zip)

argilo, 2025-12-23:
I had a quick look at the code, and it appears that when an update is made via TCP, the encoder rewrites the RDS groups immediately, even if it's in the middle of streaming out a group. I imagine this could result in a broken mixture of data for a brief time. The encoder should probably finish streaming the current groups before changing over to the new ones, rather than abruptly switching over mid-group.

gabba43, 2025-12-23:
Great. I'll see what I can do to ensure that! I'll post my solution then. Merry christmas to everyone, it's already the 24th here.