bastibl/gr-rds #93: rds_tx.grc doubt

CarlosPisco, 2025-01-19:
Hi there, to probe the tx of RDS with gnuradio what SDR device could I use?
I think that that file only transmits that data in RDS, but then how i make sure it was done successfully?  Do I need an actual radio that displays that info?
Could I use an adalm pluto?

Thanks in advance

bastibl, 2025-01-20:
You can use any SDR that is supported by GNU Radio (which is basically every SDR) and that supports the required frequency and bandwidth. AFAIS, the Pluto only goes down to 325 MHz (https://www.analog.com/en/resources/evaluation-hardware-and-software/evaluation-boards-kits/adalm-pluto.html#eb-overview), while broadcast radio is typically around 100MHz.

CarlosPisco, 2025-01-28:
Thanks for your response, as I am new in this field I struggle sometimes to get some ideas. Do you know where I can find the source code of the block rds_encoder? I mean, your code works great but I would like to understand the idea behind this block.

bastibl, 2025-01-28:
Most blocks are implemented in C++. The code for these blocks is in the `lib` folder. For the encoder it's: https://github.com/bastibl/gr-rds/blob/maint-3.10/lib/encoder_impl.cc

In this module, only the GUI Panel is implemented in Python. See the `python` folder for the code.

You can also have a look at the GNU Radio Wiki for general information, e.g.: https://wiki.gnuradio.org/index.php/OutOfTreeModules

CarlosPisco, 2025-02-05:
Hi there, I've been checking the code and stuff, I saw that there is an standard which you are referring to, would you give the name of it to get a better understanding of your program?

Thanks in advance

bastibl, 2025-02-06:
This repo is about FM RDS. See IEC 62106, for example.

CarlosPisco, 2025-02-13:
Hi there, thanks for your responses, it really helped me get through your project. 
So far I'm also investigating related to this, so do you know any projects where I can transmit digital media using SDRs as tx and rx?, besides that I was wondering if you also know some projects related to this one but regarding digital radio.

Thanks in advance

bastibl, 2025-02-17:
I'm not sure what you are asking for, but this website lists GNU Radio modules for different technologies: https://www.cgran.org/. Maybe you'll find something there. In addition to that, there are OFDM and DTV examples that come as part of GNU Radio.

CarlosPisco, 2025-03-14:
Hi there, thanks for your help. I was trying to make some changes in the tx, like changing the 57kHz for rds  for a higher frequency and doing something similar with the bw (making it wider) for the rds data. The thing is that by doing something similar in the receiver where I set the other frequency instead of the 57kHz, and also the bw, but it seems to not be enough to decode the data. It keeps synchronizing and desynchronizing, so what other changes would you suggest in the rx to make this work? Does it also has to do with the code of the .cc files you created? 

Hope four your response.

bastibl, 2025-03-18:
There are a lot of parameters in the flowgraph. The ones you mentioned are the main ones, but it's hard to tell what might be the problem. I'd just add graphical outputs and check, if the signal looks as expected at the different stages of the flowgraph. Also don't change the center frequency too much. At some point there are issues with the baseband sample rate etc. But there should be plenty of headroom.

The RDS-specific modules are only used at bit-level. I don't think that they are the issue here.

CarlosPisco, 2025-04-22:
HI there, so in the meantime i was making some changes and stuff, the thing is that i get this error with the symbols from time to time, idk if you a clue what may be causing it, as I told you before it gets SYNC and SYNC after a period of time. Thanks in advance.

[image]

argilo, 2025-04-26:
I believe I have seen that behaviour on rare occasions. It can occur if the Symbol Sync block happens to lock on halfway between two Manchester-encoded bits. The symbol synchronization in rds_rx.grc could probably be improved.