bastibl/gr-rds #94: Add ability to use raw hex pi code

pclov3r, 2025-02-25:
The current method of generating a PI code does not really work for the US. The US uses encoding method that maps a PI code to a call sign. 

A example of a calculator for this is at https://db.wtfda.org/rds2.html

It would be nice if we could disable the PI country code, PI coverage Area, and PI program Reference Number and just input a raw hex pi code.

bastibl, 2025-02-27:
We could add a PI Country Code `Other` (or something like that) that just uses the raw reference number. Would this be fine for you?  

Via live reconfiguration through the message port you can already set the PI via a hex number: https://github.com/bastibl/gr-rds/blob/maint-3.10/lib/encoder_impl.cc#L194-L198

pclov3r, 2025-02-28:
That sounds like it should be good!

Although, It appears currently it expects 3 characters so with "Other" would it allow 4 so you can put any Pi code in there you wanted? 

Thanks

bastibl, 2025-02-28:
Could you give this PR a try: https://github.com/bastibl/gr-rds/pull/96?

commit 12f21188 (2025-02-28): encoder: allow setting raw pi code

fixes #94
closed 2025-03-02 by argilo