bastibl/gr-rds #95: Forced to use AF for RDS encoder

pclov3r, 2025-02-25:
It appears you are forced to use a AF for the RDS encoder. 

It would be nice if you could disable this.

bastibl, 2025-02-27:
I think, doing this properly would require larger changes to the encoder logic. If there are no alternative frequencies, one would typically send other blocks. And there could be more than one alternative frequency. I think the block should ideally accept a vector of alternative frequencies, but I do not have the time to implement this at the moment. Feel free to open a PR.

pclov3r, 2025-06-11:
> I think, doing this properly would require larger changes to the encoder logic. If there are no alternative frequencies, one would typically send other blocks. And there could be more than one [...]

To add to this it looks like trying to implement this would be a big task. It appears that the current way Group 0A is sent might be a problem with this. It actually as of right now it's perhaps not complaint and might have issues with some receivers.