bastibl/gr-rds #97: Something weird is going on with TP / TP set to false doesn't disable TP

pclov3r, 2025-03-02:
Not sure what's going on with TP but it appears there is some weird behavior going on now.

Testing https://github.com/bastibl/gr-rds/pull/96 more this is what I'm observing

I'm using the PI Country Code of "Other" 

TP set to "False" or "True" doesn't disable it. The TP group is sent anyways? 

With TP set to "False" it appears some RDS Encoder will say it belongs to random European country based on the PI code set. With it set to "True" That goes away.

Not sure what's going on here?

argilo, 2025-03-02:
What PI value are you using?

Are you sure that SDR Console's decoding is correct? I had a look at how gr-rds encodes TP, and it appears to match the spec.

pclov3r, 2025-03-02:
Anyhow it seems like the PI Country code of "Other" isn't actually doing anything? 

If I use a PI Country Code of anything besides "Other" I see the exact same behavior with TP.  

In this case I was using 0xFFFF and it was reporting Egypt.

argilo, 2025-03-03:
I tested this, and it appears the TP bit is encoded correctly and has no effect on the PI value, which is also encoded correctly. I suspect the issue you're seeing is in SDR Console.

pclov3r, 2025-03-03:
Thanks for your testing!

At this point likely sounds like a bug in the SDR-Console RDS decoder. I'll go ahead and close this off. I'll re-open it if something changes.

closed 2025-03-03 by pclov3r